IanHarvey/bluepy #353: Raspberry Pi Camera and BTLE Conflict

hteng1995, 2019-03-22:
Hello

I am having an issue when implementing bluepy while also taking photos using the Raspberry Pi Camera (using the python library picamera) on my Raspberry Pi 3B+. In my code, I first scan and connect to a designated bluetooth device, after which I take a photo. After I take this photo, the Raspberry Pi seems to be no longer able to receive messages/notifications from the previously connected peripheral. In addition, when that program completes and returns, I cannot re-scan for more devices when I restart the program, with this error resulting: 

[image]

I am then able to eliminate this error by power-cycling the Raspberry Pi. However, then the same issue then arises when I run the program. FYI, I am using python's Threading module to create a daemon that runs BLE in a separate thread while the main thread does other tasks like taking photos. 

Any feedback or insight into this problem would be very helpful.

Firefox2005, 2019-05-04:
Did you find a solution for that? I am trying to use the camera while doing bluetooth work and am experiencing exactly the same.

Firefox2005, 2019-05-04:
Can be closed. It works fine with an external bluetooth adapter - so this seems to be a hardware related topic.