IanHarvey/bluepy #354: waitForNotifications can return None while data is pending

Daniel-Christian-CardinalPeak, 2019-03-22:
waitForNotifications ends up in _waitResp to do the actual waiting on and reading from the helper process.

If more than one line was ready (due to packets coming in faster than being polled for), _waitResp only returns the first line.

On the next call, _waitResp will call poll().  If there isn't additional data (to what was already read, but not returned before), poll() returns None and _waitResp returns None.

I have a peripheral that returns a fast burst of packets and then goes quiet.  That triggers this every time.  Unfortunately, I can't share the code (and it's super complicated anyway).

You need readline() to be able to check for pending data without blocking, but it won't do that.  What I've done in the past is to create a Queue and a Thread.  The thread just blocks in readline() and queues every line.  You can then check for an empty queue or call queue.get() with a timeout.  There is more approaches here: https://stackoverflow.com/questions/375427/non-blocking-read-on-a-subprocess-pipe-in-python

Daniel-Christian-CardinalPeak, 2019-03-22:
Here is a patch that fixes the problem.  I can do a pull request if you want, but that takes longer.

[bluepy.txt](https://github.com/IanHarvey/bluepy/files/2997585/bluepy.txt)

sonologic, 2019-12-05:
Hi, so after a roller-coaster ride debugging session, I reached the same conclusion and wanted to create an issue. When searching before creating a new issue, found this issue, and the patch. Thanks! Has anyone done a PR for this one already? If not, I can probably pick this up.