IanHarvey/bluepy #358: Have bluepy support for Yocto

dhruvkakadiya, 2019-04-10:
There is no direct OpenEmbedded support for `bluepy` library to be included as part of Yocto recipe.

csonsino, 2019-05-10:
I started this @ https://github.com/csonsino/meta-bluepy

Will do some further testing on dependencies, and would ultimately like to submit the bluepy recipes as a PR to openembedded/meta-openembedded.

jajones8710, 2019-06-24:
This is the recipe that I am using and it is working
`DESCRIPTION = "BLE interface library"
HOMEPAGE = "https://github.com/IanHarvey/bluepy"
SECTION = "devel/python"

DEPENDS = "${PYTHON_PN} glib-2.0"
RDEPENDS_${PN} = "bluez5"

LICENSE = "BSD-3-Clause"
LIC_FILES_CHKSUM = "file://${COMMON_LICENSE_DIR}/BSD-3-Clause;md5=550794465ba0ec5312d6919e203a55f9"

inherit setuptools3

SRC_URI = "git://github.com/IanHarvey/bluepy.git;protocol=git;rev=53ce2f2388a936663b94f00636cc2e3677325182"

S = "${WORKDIR}/git"

TARGET_CC_ARCH += "${LDFLAGS}"`

I know the license is not correct and will have to be fixed.