IanHarvey/bluepy #364: Python 3.7.3

bachoo786, 2019-06-27:
Hi

I am trying to use bluepy in one of my docker containers i.e. homeassistant but ever since it got upgraded to python 3.7.3, bluepy has stopped working i.e. connecting and controlling my BLE devices.

Can anyone please help me out?

Thanks.

strongwillow, 2019-06-28:
I have the same issue.

> 
> Unhandled exception in thread started by <function waitForNotifications at 0xb6086db0>
> Traceback (most recent call last):
>   File "/home/pi/XXX/Aidlab/linux.py", line 75, in waitForNotifications
>     if peripheral.waitForNotifications(1.0):
>   File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 560, in waitForNotifications
>     resp = self._getResp(['ntfy','ind'], timeout)
>   File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 407, in _getResp
>     resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
>   File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 338, in _waitResp
>     if self._helper.poll() is not None:
> AttributeError: 'NoneType' object has no attribute 'poll'
>

scottyphillips, 2019-07-23:
Also seeing the same errors.

furiaceka87, 2019-08-07:
As per obelix reference I had problem with Mi Flora in Home Assistant using external Bluetooth Dongle or Raspberry onbard Bluetooth.
I don't know if the issue is strictly related to bluepy.

RulerKun, 2019-10-09:
seeing same issue, i am using Raspberry Pi 4