IanHarvey/bluepy #367: Can not make executable

Machoney, 2019-08-07:
Hi. This is my first project on Linux. I do not know how "executables" works here and quick research showed to use PyInstaller like on Windows.

I'm using:
- Ubuntu 18.04 LTS with system python 3.6
- Python 3.7.4 on venv (i think this is important regarding my problem)
- Bluepy 1.3.0
- Pyinstaller 1.5

The installation went with some warnings, but no one of them concern Bluetooth or bluepy. When I run my program, have error like his:
`Traceback (most recent call last):
  File "app/Main.py", line 5, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "/home/kacper/PycharmProjects/app/venv/lib/python3.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 627, in exec_module
    exec(bytecode, module.__dict__)
  File "app/Sensor.py", line 4, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "/home/kacper/PycharmProjects/app/venv/lib/python3.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 627, in exec_module
    exec(bytecode, module.__dict__)
  File "scripts/BLE/SensorBLEController.py", line 1, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "/home/kacper/PycharmProjects/app/venv/lib/python3.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 627, in exec_module
    exec(bytecode, module.__dict__)
  File "bluepy/__init__.py", line 2, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "/home/kacper/PycharmProjects/app/venv/lib/python3.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 627, in exec_module
    exec(bytecode, module.__dict__)
  File "bluepy/btle.py", line 889, in <module>
  File "bluepy/btle.py", line 870, in __init__
  File "bluepy/btle.py", line 882, in get_json_uuid
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/_MEI6WEUtd/bluepy/uuids.json'
[32758] Failed to execute script GatewayApp
`

I went deep into the code, and it looks like the path to the bluepy-helper is incorrect.

`helperExe = os.path.join(script_path, "bluepy-helper")`
 used in:

```
def get_json_uuid():
    import json
    with open(os.path.join(script_path, 'uuids.json'),"rb") as fp:
...
```

I don't understand why he's referring to the _/tmp/_ folder, if it's generated from venv.

Did anyone have such a problem?

Machoney, 2019-08-07:
Okey, I have figure it out.

I have build bluepy from source in my venv. Then I set to run bluepy-helper as a normal user.
`sudo setcap 'cap_net_raw,cap_net_admin+eip' bluepy-helper`

After that, I have run pyinstaller one more time and it works.

closed 2019-08-07 by Machoney

bambachas, 2021-05-19:
@Machoney  can you describe with details what you done to make it works
i have the same problem

JimmyWang1989, 2022-07-14:
@bambachas this issue is caused by the required data file from bluepy is missed when PyInstaller trying collect the dependency. Put the attached hook-bluepy.py(change from .txt to .py) under <Path to pythonX.Y>/site-packages/PyInstaller/hooks and re-run the PyInstaller generation process can fix the problem
[hook-bluepy.txt](https://github.com/IanHarvey/bluepy/files/9108970/hook-bluepy.txt)
.

amartineze2i, 2023-01-30:
> @bambachas this issue is caused by the required data file from bluepy is missed when PyInstaller trying collect the dependency. Put the attached hook-bluepy.py(change from .txt to .py) under /site-packages/PyInstaller/hooks and re-run the PyInstaller generation process can fix the problem [hook-bluepy.txt](https://github.com/IanHarvey/bluepy/files/9108970/hook-bluepy.txt)

It works for me.