IanHarvey/bluepy #373: BTLEManagementError: Management not available (permissions problem?)

FunPsycho, 2019-08-31:
I'm running this repository: https://github.com/GammaGames/kano-wand-demos and am getting this error from bluepy

`Scanning...
Scanning for 1.0 seconds...
Traceback (most recent call last):
  File "callback.py", line 64, in <module>
    main()
  File "callback.py", line 19, in main
    wands = shop.scan(connect=True)
  File "/home/nynja/Wand/kano-wand-demos/kano_wand/kano_wand.py", line 669, in scan
    self._scanner.scan(timeout)
  File "/usr/local/lib/python3.7/dist-packages/bluepy-1.3.0-py3.7.egg/bluepy/btle.py", line 852, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python3.7/dist-packages/bluepy-1.3.0-py3.7.egg/bluepy/btle.py", line 790, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python3.7/dist-packages/bluepy-1.3.0-py3.7.egg/bluepy/btle.py", line 309, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python3.7/dist-packages/bluepy-1.3.0-py3.7.egg/bluepy/btle.py", line 366, in _waitResp
    raise BTLEManagementError("Management not available (permissions problem?)", resp)
bluepy.btle.BTLEManagementError: Management not available (permissions problem?)`

So I tried to just run the bluepy-helper and when i type the scan command I get nomngt error code.  

Is there something i have to do to get permissions?

I am running Debian on Windows 10 - https://www.microsoft.com/en-us/p/debian/9msvkqc78pk6?activetab=pivot:overviewtab

I don't know if that is the issue or not, but I don't have another machine that is linux based available to me right now.

Thanks

FunPsycho, 2019-09-01:
When I run the bluepy-helper command I get this error

rsp=$errcode=$nomgmt

sterwen, 2019-09-06:
Looks a permission problem really. To perform a BLE scan you need to be root. Don't know the fundamental reason behind that, but if the bluepy-helper executable does not have root permissions it fails.

FunPsycho, 2019-09-06:
I run it with sudo and I almost am sure my user has root access.  Is there something else I need to do?

InquisitiveAS, 2019-11-04:
I am having the same issue, did someone figure out the issue

FunPsycho, 2019-11-04:
I had to log in as root and that fixed it for me.

closed 2020-02-01 by MysticNynja

teojunjie, 2021-02-13:
Any luck with this?