IanHarvey/bluepy #375: BLE Gatt MTU exchange is limited to 23 bytes

modestid, 2019-09-08:
The Raspbian Stretch Lite, Kernel Version: 4.14.52+,  BlueZ Version: 5.43 always specifies 23 bytes as the receive MTU size. When making a Bluetooth LE GATT connection both devices exchange their MTU size. The smallest value of both is used, which dictates the notification data size.  I need to increase the MTU size so I can notify with larger data packets. I've seen that the L2CAP PDU size limits the ATT mtu size, and that the patch "Change default MTU for L2CAP ATT channel" by Andre Guedes was introduced in [Linux 3.6], so I don't know why I'm still limited to 23 bytes. Does anyone have insight into this issue?

tzachi-dar, 2019-09-14:
I also have this problem with a rpii.
Please update if you find anything.

StefJar, 2019-10-21:
same for me on my Ubuntu.
Everyone that read this [article](https://punchthrough.com/maximizing-ble-throughput-part-2-use-larger-att-mtu-2/)  tends to set the MTU to his needs

sterwen, 2019-10-21:
On my side we are still investigating. We have but traces deep in the Bluez stack part of the bluepy-helper, and so far detected nothing wrong in there. Our investigations are now in the direction of the HCI layer and BLE Chip (TI), but this is not easy.

tzachi-dar, 2019-10-21:
I'm also working on debugging the bluez code.
Maybe we should open an issue there and see if they can help?

schneider42, 2020-04-18:
I'm observing a similar issue with another device (card10 badge). Did anyone of you make progress?

tzachi-dar, 2020-09-22:
This was solved on https://github.com/IanHarvey/bluepy/commit/d6f425a11f68b7dd51ca7ca44c56a30e9716b6b4

To anyone getting here by google:
What one should do is follow instructions on https://github.com/IanHarvey/bluepy on how to install bluepi from source.

Two things to note:
1) Make sure that your bluez version is at least 5.47 (use "bluetoothctl -v" to know what your version is).
2) if you want to use it with python 3, make sure to run python3 in places that the instructions say python.

Jurpp, 2021-06-02:
As this issue is still open I would like to confirm that after manually building Bluepy from source my Raspberry Pi automatically negotiates an MTU of 247. Maybe we can make a new release containing this fix?

tzachi-dar, 2021-06-02:
The issue is solved, but one needs to build it themselves.
+1 for a new release.

grosdode, 2021-06-24:
I tried to build it (clean pi 4):
```
sudo apt update
sudo apt full-upgrade
reboot
sudo apt-get install git build-essential libglib2.0-dev
git clone https://github.com/IanHarvey/bluepy.git
cd bluepy
python setup.py build
sudo python setup.py install
```
but I get an error “ImportError: cannot import name btle”. (I never build something like this before by myself)

lauraece, 2023-02-21:
Any update to change MTU and see more than 20 bytes?

Thnks.

Ballen7, 2023-03-17:
I can also confirm that commit https://github.com/IanHarvey/bluepy/commit/d6f425a11f68b7dd51ca7ca44c56a30e9716b6b4 fixed this issue.

@grosdode delete the bluepy folder e.g. 
`sudo rm -rf bluepy/bluepy`