IanHarvey/bluepy #378: Unable to get notifications

aguileraGit, 2019-09-28:
I don't ever receive notifications from Characteristic 6E40000**6**-B5A3-F393-E0A9-E50E24DCCA9E. I must enable the sensor by writing 0x01 to Characteristic 6E40000**9**-B5A3-F393-E0A9-E50E24DCCA9E.

Verified Services and Characteristics using LightBlue iOS App. Log:
```
21:58:33.956 — Characteristic (6E400009-B5A3-F393-E0A9-E50E24DCCA9E) wrote new value: <01>
21:58:33.999 — Characteristic (6E400009-B5A3-F393-E0A9-E50E24DCCA9E) read: (null)
21:58:39.339 — Characteristic (6E400006-B5A3-F393-E0A9-E50E24DCCA9E) notified: <06020341 090000>
21:58:39.849 — Characteristic (6E400006-B5A3-F393-E0A9-E50E24DCCA9E) notified: <06020338 090000>
21:58:40.329 — Characteristic (6E400006-B5A3-F393-E0A9-E50E24DCCA9E) notified: <0602033f 090000>
21:58:40.840 — Characteristic (6E400006-B5A3-F393-E0A9-E50E24DCCA9E) notified: <06020338 090000>
21:58:41.349 — Characteristic (6E400006-B5A3-F393-E0A9-E50E24DCCA9E) notified: <0602033a 090000>
```

My Python App:
```python
from bluepy.btle import UUID, Peripheral, DefaultDelegate
import time
import struct

class MyDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleNotification(self, cHandle, data):
        print(cHandle, data)
        #Should recieve notification from 6e400006-b5a3-f393-e0a9-e50e24dcca9e

#Connect to device
print('Connecting...')    
dev = Peripheral("df:4c:cf:e7:9b:1f")

#Setup Notifications
dev.setDelegate( MyDelegate() )

#Get sensor services
sensorServices = dev.getServiceByUUID( UUID("6E400000-B5A3-F393-E0A9-E50E24DCCA9E") )
print(sensorServices)

#Get Stream Characteristic
enableStreamCharacteristic = sensorServices.getCharacteristics( UUID('6E400009-B5A3-F393-E0A9-E50E24DCCA9E') )[0]
print(enableStreamCharacteristic)

#Write 0x01 to 6e400009-b5a3-f393-e0a9-e50e24dcca9e
enableStreamCharacteristic.write( struct.pack('<B', 0x01) )

print("Waiting...")
while True:

    if dev.waitForNotifications(1.0):
        # handleNotification() was called
        continue

dev.disconnect()
print('Done')
```

Results:
```
Connecting...
Service <uuid=6e400000-b5a3-f393-e0a9-e50e24dcca9e handleStart=9 handleEnd=65535>
Characteristic <6e400009-b5a3-f393-e0a9-e50e24dcca9e>
Waiting...
```

shadetree01010100, 2019-10-21:
If you read that characteristic after writing, does it reflect the value that you wrote? In your log it looks like the value read was `null`.

sookah, 2020-05-21:
@aguileraGit you have to write the `CCCD` value to the descriptor not the characteristic. You can do that by getting the `handle` for the characteristics and adding 1 to it to get the descriptor handle. And use the write command to send the 0x01 value. 

```
# Descriptor update to CCCD and enable task        self.p.writeCharacteristic(self.task_characteristic_handle+1, b"\x01\x00", True)
```