IanHarvey/bluepy #379: Failed to execute management command 'scan'

JackHilliard, 2019-10-08:
I am running bluepy through a script on a Raspberry Pi running Raspbian Jessie.
I can scan every second for anywhere between a few seconds and nearly a minute until I get the error:

File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py" , line 852, in scan
self.start(passive=passive)
File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py" , line 800, in start
self._mgmtCmd(self._cmd())
File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py" , line 312, in _mgmtCmd
raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagamentError: Failed to execute management command 'scan'

I've had this code working on other devices but I have no clue why it doesn't work on this pi.

Any help is much appreciated

JackHilliard, 2019-10-08:
I reinstalled bluepy and changed the scan interval to 2 seconds and it seems to be working fine now

RulerKun, 2019-10-09:
Hi @JackHilliard how did you do that?

JackHilliard, 2019-10-09:
To reinstall bluepy:
sudo pip3 unintall bluepy
sudo pip3 isntall bluepy

(Reinstalling only worked for python3 not python2.7 for this device)

I changed the scan interval by changing the timeout when calling scanner.scan

JackHilliard, 2019-10-09:
Turns out that only worked temporarily, I now get the error again after 1 minute of using bluepy

shadetree01010100, 2019-10-21:
I assume that you're not calling `scan()` more frequently than the `timeout`, and/or from multiple threads, right? It might help if you can post the code in which you're looping and calling `scan()`

summer2009, 2020-04-01:
I have almost the same problem.The code like this:
for i in range(1000):
      scan(,10)...