IanHarvey/bluepy #383: Use a different bluez

alcir, 2019-10-31:
Hello.
I would like to package bluepy for Fedora.
Fedora doesn't allow to bundle a release of bluez in the package. It should use the system release of bluez.

There is a way to provide a method to compile bluepy using the system version of bluez?

parkerlreed, 2019-10-31:
I do see where the Arch package cleans up a bluez-src. It seems just removing that is enough.

alcir, 2019-10-31:
Sorry but since I'm not too practical with sources etc. my terminology could be incorrect.

I mean. In order do build the python package, bluepy doesn't use C headers and so on provided by system, but the source package includes bluez sources, and the Makefile points to them.
Or I'm wrong?

parkerlreed, 2019-10-31:
For that I'm not sure. I've had no issues with with the packaging on Arch at least. bluez updates and everything is fine.

IanHarvey, 2019-11-02:
bluepy uses a copy of some source files from the BlueZ project to build a C language executable, `bluepy-helper`, which is a shim between Python and the kernel's Bluetooth interface.  It does not build or install any binaries which are the _same_ as what the BlueZ package provides - so at no point would you have a "bluepy" and a "system" version of the same thing on your disk.

Note that there isn't a realistic way to "use the system's BlueZ" - BlueZ itself provides various command line executables and daemons, but there isn't a library or low-level interface suitable for calling from Python. (I looked at the D-Bus interface but at the time it was very incomplete, and its availability is likely to vary a lot between distributions).

brianjmurrell, 2019-11-06:
> but there isn't a library or low-level interface suitable for calling from Python

The `libbluetooth.so` library and associated headers that is packaged in [bluez-libs-devel](https://src.fedoraproject.org/rpms/bluez/blob/master/f/bluez.spec#_263) (on Fedora/RHEL) is not suitable?