IanHarvey/bluepy #386: What means "Not Supported"?

TuChucheng, 2019-11-12:
I have tried "sudo" :
cd /usr/local/lib/python3.5/dist-packages/bluepy
sudo ./bluepy-helper 0
then I typed "le on" or "scan"
but I got this:
rsp=$mgmtcode=$mgmterrestat=hCemsg='Not Supported

What means "Not Supported"? And, how can I fix it?

TuChucheng, 2019-11-12:
For more info : My PC environment is
Host : Win10
VMware Workstation 15 player : Ubuntu 14.04

Thanks in advance!

mdxs, 2020-10-21:
Unfortunately this will not work properly, as noted in https://kb.vmware.com/s/article/2005315 there are severe limitations with using Bluetooth/BLE inside a guest VM. I've had similar issues with a Xubuntu guest VM, where parts may have seem to work some of the time or partially; though in the end it always failed.

This is thus not an issue with BluePy, just a limitation of Virtual Machines (at least with VMware).