IanHarvey/bluepy #389: Fail to choose interface when creating connection on arm64

spotperg, 2019-12-03:
Hi,
I am using bluepy with bluez on an arm board. I use two nordic dev kits with the zephyr HCI_UART sample code running, connected via USB.The board has no native bt device. I use btattach to bring up these two boards, and get access to devices HCI0 and HCI1. So far, so good. When I scan I can use the iface param to set which board I want to use. Using btmon I can verify that this is working as I expect. 
However, when I try to create the Peripheral object, setting the iface param fails. btmon tells me that the device I attached last is the one being used. I.e. if I execute btattach -B /dev/ttyACM0 and then the same with ttyACM1, hci1 will be used. If I kill the first instance of btattach, and the re-start btattach -B /dev/ttyACM0, hci0 will be the prefered. In my python script I print the iface-property available in the Peripheral object. It is consistent with what I request, and not with what btmon tells me. I've also verified that this happens with bluepy-helper and btmon. I've attached three screenshots of bluepy-helper, showing that hci0 is always used.

When testing on my PC (native bt + 1 nordic kit) this was not an issue. Then I was able to create two Peripheral connections on two separate interfaces, just as I expected.

I have had no issues with bluepy/bluez on the arm board, when using one interface. 

Any help / follow-up question / stuff to investigate will be appreciated. 

Versions on arm64:
Bluepy: 1.3.0
Bluez: 5.50.0-1
Kernel: 4.19 (compiled by me)
Python 3.7.3
My PC runs bluez 5.50-ubuntu4 and kernel 5.3.0-23 stock ubuntu. Bluepy 1.3.0. Python 3.7.5.
[image]
[image]
[image]

kg-meso, 2023-02-13:
I am having the same issue, did you find a fix to this? #501

peran1337, 2023-02-14:
Honestly, I can't remember :smiley: