IanHarvey/bluepy #394: Getting RSSI value from a connected device

marquesn, 2020-01-17:
Hi all,

I can get the RSSI values from BLE devices which are not connected.

However, after I connect a device using Peripheral.connect(dev1), the Scanner() keeps returning the same RSSI value it had for 'dev1' before the connection (basically the RSSI is not being updated for connected device dev1).

Has anyone come across this before?

Thanks

mmaziesf, 2020-01-24:
Same here. I wanted to measure RSSI values from an active BLE connection and couldn't find any method in bluepy to do this. Please advise if you could find a solution for this. Thanks.

marquesn, 2020-01-28:
I found a work-around but it wasn't using bluepy, or any other library in the raspberry pi. Unfortunately I have not found anything from the RPi side that would do this.

What I did was:

On the BLE device I'm using, I programmed it to get the RSSI of the active connection to write it to a BLE characteristic. The RPi then reads this characteristic periodically, thus obtaining the RSSI value.

Another example was by using a ESP32 as the BLE device, I simply put it to continue advertising after a connection is established. This way, the RPi would still see its RSSI after connecting.

coofercat, 2023-01-04:
I wouldn't call myself an expert, but I _think_ that once you "connect" to a BLE device, it'll stop broadcasting (although I guess this is vendor/model specific). As such, the `Scanner()` won't actually be receiving any further information, so can't update the RSSI value.

Intuitively, it should be possible to connect to a BLE device, and then periodically ask it to respond (sort of like an ICMP ping). The response (which can contain no real data) will be received with a signal strength which the server ought to be able to determine and report to the application. This is what the cheesy ifindu app that came with my tags seems to be able to do on my phone (as it gets more frequent updates when "searching" for a tag).

Either way, I can't seem to see any RSSI type information in anything bluepy provides to me (once connected). I don't know if that's a bluepy issue, a helper issue or just the way BLE works...?

coofercat, 2023-01-05:
I wouldn't call myself an expert, but I _think_ that once you "connect" to a BLE device, it'll stop broadcasting (although I guess this is vendor/model specific). As such, the `Scanner()` won't actually be receiving any further information, so can't update the RSSI value.

Intuitively, it should be possible to connect to a BLE device, and then periodically ask it to respond (sort of like an ICMP ping). The response (which can contain no real data) will be received with a signal strength which the server ought to be able to determine and report to the application. This is what the cheesy ifindu app that came with my tags seems to be able to do on my phone (as it gets more frequent updates when "searching" for a tag, which requires "connection").

For me the question is "how can I get RSSI information from a connected beacon device?"