IanHarvey/bluepy #396: Connection problem

jramonlab, 2020-01-24:
Hello All,

After installing bluepy in my RPI3 (to be central) and getting started with it I have detected connection issues:
Either the connection is not done or it is inmediately (2-5 sec.) disconnected after succesfull connection.
I am trying to connect to an ESP32 device (to be peripheral), which properly gets connected with an Android Smartphone.

The devices get properly connected via hcitool, but when trying with gatttool the problem is as described: no success or disconnection right after connection.

I have done a lot of search and after updating the BlueZ from **5.43 to 5.48** the issue remains.
It may be sth related to timeouts, but not sure. I guess it may also be sth related to versions incompatibility or sth in my system that is not properly configured.

Find below description  of my system:
- bluepy 1.3.0
- Python 3.5.3
- gatttool
- hcitool 5.43
- Bluetoothd 5.43
- RPI3 : Linux kernel 4.19.46-v7+
- blueZ Version: 5.43-2+rpt2+deb9u2 (updated to 5.48)

Any advice will be welcome.

Regards, Ramon.

jramonlab, 2020-01-24:
I have noticed that although I have updated BlueZ from 5.43 to 5.48 the tools Bluetoothd and hcitool are still 5.43. Is that normal?

CLDiego, 2020-02-16:
Hi there, would you be able to share some working code? I've tried to use an RPI3 B+ with bleepy and while I can connect to devices and get some notifications, there are some data losses. There appears to be a bug in the RPI firmware that makes BLE unstable. 

So if you're trying to do something really complex with it I would advise you no to use an RPI until they fix the firmware issue. The RP4 uses the same wifi/Bluetooth chip so I assume it's gonna present the same problem.

I also tried using a USB dongle with the same data losses and disconnection problems.

jramonlab, 2020-02-18:
Hi

Thanks for the comment. Unfortunately there is no code to share. I stopped using BLE/BT and changed to use my ESP32 directly via WiFi.

Do you know where I can follow how BLE is fixed on RPI?

Regards, Ramon.

CLDiego, 2020-02-18:
No problem. You can find see the current [issue](https://github.com/raspberrypi/linux/issues/1548) here. 

Hopefully, they will fix it soon.

RulerKun, 2020-07-15:
@CLDiego I've tried my RPI 4 and getting some problems with the notifications too

CLDiego, 2020-07-15:
Hi @RulerKun, sadly not being able to determine the real issue with the loss of data on the PI I decided to switch to an ESP32 to handle my BLE data and then transmit it over wifi to the PI.

Playing more with BLE I can make some suggestions that might help:

1. Play around with the connection interval of your Peripheral. Remember that the Central is in charge of choosing the best connection characteristics for him. But you can force it by defining your min and max connection parameters for the Peripheral.  I never tried this but I'm assuming that the PI might have a harder time keeping alive a high connection interval value (e.g. 3-4 seconds) so try to go lower.

2. As I mentioned, it seems that the problem is highly related to the silicon used in the RPI 3B+ and RPI4. I know that the RPI zero uses a different chip, so you might have more luck trying to use one.

3. The RSSI is quite important when using notifications. I got the best resuts with an RSSI > -70dB. Lower values, give the worst results or the peripheral does not connect at all. I'm not sure what you are using as a Peripheral but it's worth noting that some BLE modules have a Tx power of 0 dB meaning that you really need to be close to your Central, in this case your PI.

4. Try using indications instead of notifications. This will ensure you get all or most of the data from your Peripheral. However, this did not work for me because I would sometimes get data in the wrong order, so you might need to work out a way to guard your data so you can put it in the appropiate order if needed.

Hope that helps!

RulerKun, 2020-07-16:
thank you very much for your suggestions!!