IanHarvey/bluepy #404: Install on macOS fail with "fatal error: 'endian.h' file not found"

Hacker-CB, 2020-04-08:
**MacOS**: 10.15.4 (19E266)
**Xcode**: Version 11.4 (11E146)
Python 3.7.3

`pip install bluepy`

```
Collecting bluepy
  Using cached bluepy-1.3.0.tar.gz (217 kB)
Building wheels for collected packages: bluepy
  Building wheel for bluepy (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/pavel/projects/home-assistant/core/.venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy/setup.py'"'"'; __file__='"'"'/private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-wheel-7775epou
       cwd: /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy/
  Complete output (11 lines):
  running bdist_wheel
  running build
  running build_py
  Working dir is /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy
  execute make -C ./bluepy clean
  execute make -C bluepy -j1
  Failed to compile bluepy-helper. Exiting install.
  Command was 'make -C bluepy -j1' in /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy
  Return code was 2
  Output was:
  b"tar xzf bluez-src.tgz\ntouch ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c\ncc -L. -g -Wall  -Os -DHAVE_CONFIG_H -I./bluez-5.47/attrib -I./bluez-5.47 -I./bluez-5.47/lib -I./bluez-5.47/src -I./bluez-5.47/gdbus -I./bluez-5.47/btio -I./bluez-5.47/sys -I/usr/local/Cellar/glib/2.64.1_1/include/glib-2.0 -I/usr/local/Cellar/glib/2.64.1_1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.44/include -o bluepy-helper bluepy-helper.c ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c -L/usr/local/Cellar/glib/2.64.1_1/lib -L/usr/local/opt/gettext/lib -lglib-2.0 -lintl\nIn file included from bluepy-helper.c:36:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/lib/bluetooth.c:38:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/lib/hci.c:44:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/lib/sdp.c:45:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/lib/uuid.c:33:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/attrib/att.c:35:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/attrib/gatt.c:34:\nIn file included from ./bluez-5.47/lib/sdp.h:35:\n./bluez-5.47/sys/bluetooth/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/attrib/gattrib.c:36:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/attrib/utils.c:32:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/btio/btio.c:39:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/src/log.c:40:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/src/shared/mgmt.c:33:\n./bluez-5.47/lib/bluetooth.h:36:10: fatal error: 'endian.h' file not found\n#include <endian.h>\n         ^~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/src/shared/crypto.c:33:\n./bluez-5.47/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^~~~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/src/shared/att.c:34:\n./bluez-5.47/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^~~~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/src/shared/queue.c:28:\n./bluez-5.47/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^~~~~~~~~~~~\n1 error generated.\nIn file included from ./bluez-5.47/src/shared/util.c:38:\n./bluez-5.47/src/shared/util.h:27:10: fatal error: 'byteswap.h' file not found\n#include <byteswap.h>\n         ^~~~~~~~~~~~\n1 error generated.\nmake: *** [bluepy-helper] Error 1\n"
  ----------------------------------------
  ERROR: Failed building wheel for bluepy
  Running setup.py clean for bluepy
Failed to build bluepy
Installing collected packages: bluepy
    Running setup.py install for bluepy ... error
    ERROR: Command errored out with exit status 1:
     command: /Users/pavel/projects/home-assistant/core/.venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy/setup.py'"'"'; __file__='"'"'/private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-record-xm1xtsh4/install-record.txt --single-version-externally-managed --compile --install-headers /Users/pavel/projects/home-assistant/core/.venv/include/site/python3.7/bluepy
         cwd: /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy/
    Complete output (11 lines):
    running install
    running build
    running build_py
    Working dir is /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy
    execute make -C ./bluepy clean
    execute make -C bluepy -j1
    Failed to compile bluepy-helper. Exiting install.
    Command was 'make -C bluepy -j1' in /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-install-9_y6rz3f/bluepy
    Return code was 2
    Output was:
    b"tar xzf bluez-src.tgz\ntouch ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c\ncc -L. -g -Wall  -Os -DHAVE_CONFIG_H -I./bluez-5.47/attrib -I./bluez-5.47 -I./bluez-5.47/lib -I./bluez-5.47/src -I./bluez-5.47/gdbus -I./bluez-5.47/btio -I./bluez-5.47/sys -I/usr/local/Cellar/glib/2.64.1_1/inclu [... cut ...] p-install-9_y6rz3f/bluepy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/8s/tyq_h6ss1q37yh8dtypy954c0000gn/T/pip-record-xm1xtsh4/install-record.txt --single-version-externally-managed --compile --install-headers /Users/pavel/projects/home-assistant/core/.venv/include/site/python3.7/bluepy Check the logs for full command output.

```

sneko, 2020-07-29:
@Hacker-CB , it seems I have exactly the same error.

Did you find a solution to install it? Thank you,

sneko, 2020-07-29:
Ok, solved on my side.

Solution:
Since the v1.3.0 provided by "pip" on any other package manager is old from 2 years now, you should try to add the library with it's lastest commit.

So instead of doing:
`poetry add bluepy`
Do:
`poetry add git+https://github.com/IanHarvey/bluepy`

Enjoy 🎆