IanHarvey/bluepy #405: Question about the scanning channel(s)

ececli, 2020-04-30:
Hi, I want to ask a quick question here. When running the `Scanner.sccan(Timeout)`, will the device scan all 3 advertising channels in the duration `Timeout`, or it only scans one channel? If it scans 3 channels, does it scan each channel using `Timeout/3`?

Thank you!

Chang

lauraflu, 2021-02-05:
Hello! I am also interested in this question, did you find out the answer? Also, do you know if it's possible to get the channel on which a beacon was received? Thanks!

ececli, 2021-02-08:
Hi lauraflu,

I didn't get the answer from this site. But I do find a way to get the channel information in my project (You can view it on my GitHub). Basically, when the transmitter sends a signal, it sends on a particular channel. And I add the channel number information in the signal. The receiver can decode the signal and then knows the channel.

lauraflu, 2021-02-09:
Hi ececli, thank you for your answer. This seems to be the best method, as from some searches I saw that there's basically no way to get the channel the message was received on (so on the receiver side) from the driver.

KaraCemal, 2021-04-01:
Hi, ececli and lauraflu. I am also dealing with the same issue. Is there a way to a scan only one channel (37 for example)? I am asking this because I could not able to modify the channel of my BLE Advertising device (which is HM10).