IanHarvey/bluepy #412: Problems when trying to set capabilities

ctrl-shift-esc, 2020-06-19:
I am using Debian Buster on HummingBoard Pulse (i.MX8 - ARM). `uname -a` yields following result: `Linux sr-imx8 4.19.72-imx8-sr #1 SMP PREEMPT Mon Feb 10 03:59:35 UTC 2020 aarch64 GNU/Linux`.

I have installed bluepy and encountered the issue #313 and similar to #373. I tried the solution stated there but I am not able to set capabilities for the file `bluepy-helper`. I get the following error:

`debian@sr-imx8:/usr/local/lib/python3.7/dist-packages/bluepy-1.3.0-py3.7.egg/bluepy$ sudo setcap 'cap_net_raw,cap_net_admin+eip' bluepy-helper
Failed to set capabilities on file 'bluepy-helper' (Operation not supported)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file`

I have scourged for solution everywhere but could not find any. Anyone have similar issues?

mchitre, 2024-06-19:
Any luck finding a solution for the `setcap` failing? I have similar problem on Debian on Compulab iMX8m-plus

mchitre, 2024-06-19:
It seems that the Kernel is not compiled with "security labels" and hence `setcap` is not supported:
```
$ cat /proc/config.gz | gunzip | grep CONFIG_EXT4_FS_SECURITY
# CONFIG_EXT4_FS_SECURITY is not set
```