IanHarvey/bluepy #413: Check if device is already connected before connecting

gohu41, 2020-06-29:
If I create a peripheral object for a device that is already connected, it fails with a device disconnected error. 

Is there a way to check if devices are already connected and/or disconnect from devices ? 

I don't have access to the peripheral object, since the connection has been made on an another thread. 
I could do it using sub-process and hcitool but it would be cleaner using bluepy

fabocode, 2020-08-27:
I am also interested into get this kind of information. 

How did you get this information from sub-process and hcitool?

gohu41, 2020-12-28:
hcitool is a command line tool in linux: https://linux.die.net/man/1/hcitool
`hcitool scan` will give you a list of connected devices. With sub-process (https://docs.python.org/3/library/subprocess.html) you can run external commands from python, so you could probably get the information like that but it's really not ideal.

RenierM26, 2021-11-07:
Hi All,

Just import/inherit the Peripheral class and override the _connect function.

This rsp["state"][0] == "disc" or  rsp["state"][0] == "conn" seem to be the response when a device is already connected. 

The "disc" response seems to be some or other idle or busy disconnecting condition. (Still in use in other words)

This is how I'm handling it. You might need to tweak the rsp["state"][0] == "conn" response in some way.

```
    def _connect(self, timeout: int = None) -> None:
        _LOGGER.debug("Connecting to Switchbot")
        self._startHelper(self._interface)
        self._writeCmd("conn %s %s\n" % (self._mac, bluepy.btle.ADDR_TYPE_RANDOM))
        rsp = self._getResp("stat", timeout)
        if rsp is None:
            self._stopHelper()
            raise bluepy.btle.BTLEDisconnectError(
                "Timed out while trying to connect to peripheral %s" % self._mac,
                rsp,
            )
        while rsp and rsp["state"][0] in [
            "tryconn",
            "scan",
        ]:  # Wait for scan to finish.
            rsp = self._getResp("stat", timeout)

        # If operation in progress, disc is returned.
        # Bluepy helper can't handle state. Execute stop, wait and retry.
        if rsp["state"][0] == "disc":
            _LOGGER.warning("Bluepy busy, waiting before retry")
            self._stopHelper()
            time.sleep(self._scan_timeout)
            return self._connect(timeout)

        if rsp["state"][0] != "conn":
            _LOGGER.warning("Bluehelper returned unable to connect state: %s", rsp)
            self._stopHelper()
            raise bluepy.btle.BTLEDisconnectError(
                "Failed to connect to peripheral %s, rsp: %s" % (self._mac, rsp)

```