IanHarvey/bluepy #417: Notifications Example Please Correct

bapgobears, 2020-07-28:
Thanks for the great library, but I wanted to remind you the example on notifications is not correct and is causing people a lot of headaches, especially if you are new to python and/or ble.  It should be easy to correct.  Also, please make it clear that 0x02 is needed to turn on "indications" and 0x01 is for "notifications".  It's not clear they are different.  It would be great, as someone else somewhere mentioned, to add some member functions to turn notifications/indications on.  Thanks! Brett

JLiao98, 2020-11-17:
Hello, do you have an example for notifications?

CocoaBob, 2022-09-15:
I just figured it out, thanks to this answer (https://github.com/IanHarvey/bluepy/issues/378#issuecomment-632399819), we have to explicitly enable characteristic's notification with the following codes
```
myPeripheral = Peripheral("YOUR_DEVICE_MAC_ADDRESS").withDelegate(MyNotificationDelegate())
myCharacteristic = myPeripheral.getCharacteristics(uuid="CHARACTERISTIC_UUID")[0]
myPeripheral.writeCharacteristic(myCharacteristic.getHandle()+1, b"\x01\x00", True)
```
Which isn't mentioned anywhere in the doc 😂