IanHarvey/bluepy #418: Script hangs on peripheral connect function [Raspberry Pi 3 B+]

NerdboyQ, 2020-08-14:
I have been trying to build a simple test script to send bytes to my arduino's HM-10 ble module, but the script gets hung whenever I try to connect. 

My Code:
```
import time
from bluepy import *

bdaddr = "00:35:FF:0D:41:9B"
print("Connecting...")
dev = btle.Peripheral(bdaddr)
dev.connect(bdaddr)
print("Connnected")
time.sleep(10)
dev.disconnect()
print("Disconnected")
```

>NOTE: I see my HM-10 module status led changes to the connected state, but I my code never prints the "connected" string.

CLDiego, 2020-08-14:
In BLE you have to send your data through a characteristic inside a service. BLE is quite different from traditional Bluetooth, it does not talk like a simple serial communication. I have no experinece with the HM-10 Module but I assume it has some services and characteristics by default.

You can try using a BLE app on your cellphone to see the services and their characteristics. Here's a nice [article ](https://devzone.nordicsemi.com/nordic/short-range-guides/b/bluetooth-low-energy/posts/ble-services-a-beginners-tutorial)about BLE basics if you're new to it.

What your code is doing is just stablishing a connection with the device, but it does not know where to send the data inside your BLE module. Once you know to which characteristic you want to talk, you can write directly to it (bluepy has all of this functionalty). 

Hope that helps!

NerdboyQ, 2020-08-15:
@CLDiego I was about to use the Bluetooth Terminal app and send bytes to the Arduino successfully. With the script I'm able to get the characteristics, services, and descriptors. However, I'm not sure if I have to set a specific characteristics before attempting to connect or not. I see the bytes associated with each characteristic, but I'm not clear on what needed to do the same thing as the Bluetooth Terminal App.

CLDiego, 2020-08-15:
@NerdboyQ You are already connected to device, that's why you are able to read the services and characteristics. 

I didn't catch this in your orginal code. When you create the peripheral object and pass the MAC address it connects to the device. The `connect()` method is used for when you did not pass a MAC address to the object, usually you can use the peripheral object as a scanner and then use the connect method accordingly.

NerdboyQ, 2020-08-15:
Aaahhh! Ok! So I can just say `dev.write([some byte])` at that point?

CLDiego, 2020-08-15:
If you're certain that you know the GATT handle of the characteristic to which you want to write, you can use the `writeCharacteristic()` method. 
If you are not sure, you can get the service that you need using its UUID, then look for the specific characteristic using its own UUID, get it (create a characteristic object) and then write to it. Something like this:

```
service= peripheral.getServiceByUUID( service_uuid )
characteristic = service.getCharacteristics( characteristic_uuid )[0] 
characteristic.write( setup_data )
```

In the end is up to you how to implement this. If you want to avoid hardcoding handles or UUIDs you can simple get all the services and characteristics as you did before, and inside your for loops look for the specific service and characteristic.

Hope it helps.

NerdboyQ, 2020-08-15:
@CLDiego  thanks a lot! You sent me down the correct path. I'm providing my code in case anyone else comes across this issue and need clarification. This is the code I used to find which service & characteristic to use:

```
import time
from bluepy import *

bdaddr = "00:35:FF:0D:41:9B"
print("Connecting...")
dev = btle.Peripheral(bdaddr)

a = 1
print("Services...")
services = list(dev.services)
print("Services",str(services))
for service in services:
    print("attempt:"+str(a))
    print("\t\t",service.uuid,":",service)
    print("\t\t",dev.getServiceByUUID(service.uuid))
    characteristics = service.getCharacteristics()
    for c in characteristics:
        print("\t~\tCharacteristic:",c)
        c.write(b'B')
    a+=1
    time.sleep(10)

dev.disconnect()
del dev

```

Once I found the correct service & characteristics match, I used the following code to send data:

```
import time
from bluepy import *

def sendMsg(bdaddr,msg_byte):
    print("Connecting...")
    dev = btle.Peripheral(bdaddr)
    services = list(dev.services)
    characteristic = services[len(services)-1].getCharacteristics()[0]
    characteristic.write(bytes([msg_byte]))
    print("Sending message:",msg_byte)
    dev.disconnect()
    print("Disconnected.")
    del dev

bdaddr = "00:35:FF:0D:41:9B"
msg_byte = 254
sendMsg(bdaddr,msg_byte)
```

closed 2020-08-15 by NerdboyQ