IanHarvey/bluepy #420: Issue with bluepy

Vill8-0704, 2020-09-10:
Hello,
I need some help with bluepy installation.
I 'm trying to install bluepy from the command "sudo pip3 install bluepy"
but the terminal is having problems with installing. 

`Collecting bluepy
  Downloading bluepy-1.3.0.tar.gz (217 kB)
     |████████████████████████████████| 217 kB 690 kB/s 
Building wheels for collected packages: bluepy
  Building wheel for bluepy (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-97ch2306/bluepy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-97ch2306/bluepy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-9usxn23i
       cwd: /tmp/pip-install-97ch2306/bluepy/
  Complete output (11 lines):
  running bdist_wheel
  running build
  running build_py
  Working dir is /tmp/pip-install-97ch2306/bluepy
  execute make -C ./bluepy clean
  execute make -C bluepy -j1
  Failed to compile bluepy-helper. Exiting install.
  Command was 'make -C bluepy -j1' in /tmp/pip-install-97ch2306/bluepy
  Return code was 2
  Output was:
  b"make: Entering directory '/tmp/pip-install-97ch2306/bluepy/bluepy'\ntar xzf bluez-src.tgz\ntouch ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c\nmake: pkg-config: No such file or directory\nmake: pkg-config: No such file or directory\ncc -L. -g -Wall  -Os -DHAVE_CONFIG_H -I./bluez-5.47/attrib -I./bluez-5.47 -I./bluez-5.47/lib -I./bluez-5.47/src -I./bluez-5.47/gdbus -I./bluez-5.47/btio -I./bluez-5.47/sys  -o bluepy-helper bluepy-helper.c ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c \nbluepy-helper.c:33:10: fatal error: glib.h: No such file or directory\n   33 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/attrib/att.c:33:10: fatal error: glib.h: No such file or directory\n   33 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/attrib/gatt.c:32:10: fatal error: glib.h: No such file or directory\n   32 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/attrib/gattrib.c:34:10: fatal error: glib.h: No such file or directory\n   34 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/attrib/utils.c:30:10: fatal error: glib.h: No such file or directory\n   30 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/btio/btio.c:37:10: fatal error: glib.h: No such file or directory\n   37 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/src/log.c:38:10: fatal error: glib.h: No such file or directory\n   38 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/src/shared/io-glib.c:30:10: fatal error: glib.h: No such file or directory\n   30 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/src/shared/timeout-glib.c:22:10: fatal error: glib.h: No such file or directory\n   22 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\nmake: *** [Makefile:30: bluepy-helper] Error 1\nmake: Leaving directory '/tmp/pip-install-97ch2306/bluepy/bluepy'\n"                  
  ----------------------------------------                                                                        
  ERROR: Failed building wheel for bluepy
  Running setup.py clean for bluepy
Failed to build bluepy
Installing collected packages: bluepy
    Running setup.py install for bluepy ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-97ch2306/bluepy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-97ch2306/bluepy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-zojp3gmn/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.8/bluepy                                                 
         cwd: /tmp/pip-install-97ch2306/bluepy/                                                                   
    Complete output (11 lines):                                                                                   
    running install                                                                                               
    running build                                                                                                 
    running build_py                                                                                              
    Working dir is /tmp/pip-install-97ch2306/bluepy                                                               
    execute make -C ./bluepy clean                                                                                
    execute make -C bluepy -j1                                                                                    
    Failed to compile bluepy-helper. Exiting install.                                                             
    Command was 'make -C bluepy -j1' in /tmp/pip-install-97ch2306/bluepy                                          
    Return code was 2                                                                                             
    Output was:                                                                                                   
    b"make: Entering directory '/tmp/pip-install-97ch2306/bluepy/bluepy'\ntar xzf bluez-src.tgz\ntouch ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c\nmake: pkg-config: No such file or directory\nmake: pkg-config: No such file or directory\ncc -L. -g -Wall  -Os -DHAVE_CONFIG_H -I./bluez-5.47/attrib -I./bluez-5.47 -I./bluez-5.47/lib -I./bluez-5.47/src -I./bluez-5.47/gdbus -I./bluez-5.47/btio -I./bluez-5.47/sys  -o bluepy-helper bluepy-helper.c ./bluez-5.47/lib/bluetooth.c ./bluez-5.47/lib/hci.c ./bluez-5.47/lib/sdp.c ./bluez-5.47/lib/uuid.c ./bluez-5.47/attrib/att.c ./bluez-5.47/attrib/gatt.c ./bluez-5.47/attrib/gattrib.c ./bluez-5.47/attrib/utils.c ./bluez-5.47/btio/btio.c ./bluez-5.47/src/log.c ./bluez-5.47/src/shared/mgmt.c ./bluez-5.47/src/shared/crypto.c ./bluez-5.47/src/shared/att.c ./bluez-5.47/src/shared/queue.c ./bluez-5.47/src/shared/util.c ./bluez-5.47/src/shared/io-glib.c ./bluez-5.47/src/shared/timeout-glib.c \nbluepy-helper.c:33:10: fatal error: glib.h: No such file or directory\n   33 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/attrib/att.c:33:10: fatal error: glib.h: No such file or directory\n   33 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/attrib/gatt.c:32:10: fatal error: glib.h: No such file or directory\n   32 | #include <glib.h>\n      |          ^~~~~~~~\ncompilation terminated.\n./bluez-5.47/attrib/gattrib.c:34:10: fatal error: glib.h: No such file or directory\n   34 | #include <glib.h>\n      |          ^~~~~~~~ [... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-97ch2306/bluepy/setup.py'"'"'; __file__='"'"'/tmp/pip-install-97ch2306/bluepy/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-zojp3gmn/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.8/bluepy Check the logs for full command output.`

has anyone ever had this problem or has a solution?
Tnks

CLDiego, 2020-09-10:
Explaining more about your setup would help. Are you using linux? bluepy only works on linux.

The error states a problem with the wheel, have you installed the development files for the GLib library? Did you follow the installation instructions? Have you installed BlueZ?

Did you try the following?

```
$ sudo apt-get install python3-pip libglib2.0-dev
$ sudo pip3 install bluepy
```

Vill8-0704, 2020-09-10:
yes thanks all solved.

mdxs, 2020-10-21:
@Vill8-0704 As you noted that it has been solved, could you please close this issue?

JounQin, 2023-11-17:
Hi, I'm on an environment without `apt-get` available, my lib managers are `opkg` and `homebrew`, how can I install `libglib2.0-dev` via these two lib managers?

nicollelumertz, 2024-05-03:
@JounQin Hi, did you find a way? I am having the same problem