IanHarvey/bluepy #430: bluepy-helper causing 100% cpu usage

okoestner, 2020-12-03:
On my dietpi-based raspberry the bluepy-helper process causes 100% cpu usage. Other tasks can run but it the tool should be less resource consuming.

fsaris, 2021-05-04:
Probably duplicate of #332 and fixed with #446

garudaonekh, 2021-06-10:
> Probably duplicate of #332 and fixed with #446

Still the same after apply this fix

florian1984, 2021-11-03:
Any updates on this issue? Have to implement stupid workarround to fix those hangers:

```
#!/bin/bash
for PID in $(ps -ef | egrep "(/bin/sh -c /home/pi/MiTemperature2/LYWSD03MMC.py)|(bluepy-helper)" | grep -v grep | awk {'print $2'}); do
        RUNTIMESEC=$(ps -o etimes= -p "$PID")
        if [ "$RUNTIMESEC" -gt "300" ]; then kill -9 $PID; fi
done
exit 0
```

Mausy5043, 2022-03-20:
@florian1984 : thanks for that script. I was also looking for a way to automatically kill the `bluepy-helper` when it hangs.
I'm using `bluepy` v1.3.0 and it regularly hangs.
😞

Mausy5043, 2022-03-21:
I've turned the script from @florian1984 into a service. So that I don't have to monitor it myself.

The `bluepy-helper` is being killed by the service when it hangs. This results in some logging from my own application:

```
Mar 21 16:15:31 rbenvir helper-killer[3303]: ** ** bluepy-helper has been running for     331 seconds; killing 4488
Mar 21 16:15:31 rbenvir kimnaty.py[3304]: Exception ignored in: <function Peripheral.__del__ at 0x7fb159e280>
Mar 21 16:15:31 rbenvir kimnaty.py[3304]: Traceback (most recent call last):
Mar 21 16:15:31 rbenvir kimnaty.py[3304]:   File "/home/pi/.local/lib/python3.9/site-packages/bluepy/btle.py", line 630, in __del__
Mar 21 16:15:31 rbenvir kimnaty.py[3304]:     self.disconnect()
Mar 21 16:15:31 rbenvir kimnaty.py[3304]:   File "/home/pi/.local/lib/python3.9/site-packages/bluepy/btle.py", line 453, in disconnect
Mar 21 16:15:31 rbenvir kimnaty.py[3304]:     self._writeCmd("disc\n")
Mar 21 16:15:31 rbenvir kimnaty.py[3304]:   File "/home/pi/.local/lib/python3.9/site-packages/bluepy/btle.py", line 305, in _writeCmd
Mar 21 16:15:31 rbenvir kimnaty.py[3304]:     self._helper.stdin.flush()
Mar 21 16:15:31 rbenvir kimnaty.py[3304]: BrokenPipeError: [Errno 32] Broken pipe
```

Hope this helps with fixing the issue.

Mausy5043, 2022-04-12:
Might be fixed by #476.

Implemented that fix yesterday and haven't had any major bail-outs or hangs since. The occasional failure to connect is still there, but this might also be caused by the devices sometimes misbehaving.

I have now 8 Mija LYWSD03MMC devices that are polled every (approx.) 5 minutes. Data is coming in reasonably steady now. My `bluepy-helper-killer` service seems to have become redundant too. 😄  Fingers crossed.

J-Luc16, 2023-02-11:
Same issue when I moved from mijapoller to LYWSD03MMC.py :-(

Is it possible to have the killing script periodically run ?

PI3B+, Raspbian bullseye.

J-Luc

bobemoe, 2024-05-13:
I'm getting hung helpers too.

Sometimes when I call `myPeripheral.disconnect()` I get exception "Unexpected response (wr)", this leaves a hung helper.

My kill script (thanks ChatGPT) looks like this:

```
#!/bin/bash

echo "Checking bluepy-helper processes..."
# Loop through each process matching 'bluepy-helper'
ps -eo pid,comm,pcpu --sort=-pcpu | awk '$2 == "bluepy-helper" {print $1, $3}' | while read pid cpu_usage
do
  echo -n "PID: $pid, CPU Usage: $cpu_usage% - "
  if (( $(echo "$cpu_usage > 10" | bc -l) )); then
    # If CPU usage is greater than 90%, kill the process
    kill $pid
    if [ $? -eq 0 ]; then
      echo "Killed."
    else
      echo "Failed to kill."
    fi
  else
    echo "Not killed."
  fi
done
```

I run it every minute in cron:

`* * * * * /home/devuan/killbluepycpu.sh 2>&1`