IanHarvey/bluepy #441: Handling multiple bluepy-helper calls

fabocode, 2021-02-16:
Hello!

**My Setup**
- Raspberry Pi 4
- Firmware version: Linux 4.19.118-v7l+ #1311 SMP armv7l GNU/Linux
- Bluetoothd: 5.50
- bluepy: v1.3.0
 

**The Problem**
I have a program that after several hours running we checked python processes running in background with `ps -ef | grep python` and saw that we have a lot of calls made to `bluepy-helper`. I am able to end the main python script and the number of `bluepy-helper` disappears, but I think this is causing us problems in the long run, because we need the program to run continuously.

[image]

**The Scenario**
My program connects to different devices but only one at a time and is always scanning to check if there is one of this devices near to the range of connection. 

**What I tried**
I tried to restart the process with `sudo service bluetooth restart` but this only repeat the process and I will end up in the same scenario and even crash the code because ble do not work in the same way after that.

**My Question**
Is there a possibility to understand how the `bluepy-helper` works in terms of how to end a process of `bluepy-helper` ? When is `bluepy-helper` triggered to start? when is it ended? how can we manage it using the python bluepy module? or if is necessary to run external commands it will be helpful to know which ones are required.

Judas2016, 2021-04-06:
I am having a similar problem.  This only started since deploying the application on a PI4. I had it running on PI3 w/USB bluetooth for months without this happening.  I have cross checked my code and am calling disconnect on the BLE_SERVICE object and setting the BLE_SERVICE to None in finally block.  I have updated the PI4 firmware w/the latest drivers from BlueZ - same problem. https://github.com/RPi-Distro/bluez-firmware/tree/master/broadcom

doug-holtsinger, 2021-12-18:
These are the methods for starting and stopping the helper, respectively:
https://github.com/IanHarvey/bluepy/blob/master/bluepy/btle.py#L276
https://github.com/IanHarvey/bluepy/blob/master/bluepy/btle.py#L302
The same file has the calls to these two methods.  Do you have a link to your bluepy application?