IanHarvey/bluepy #464: Archive Releases?

HadetTheUndying, 2021-10-30:
Would you be open to having source archives available as releases? I would like to package this for Void Linux but without a source archive I cannot meet the quality requirements for that.

IanHarvey, 2021-11-01:
Hi - I’m very happy for you to use the source in any way you like. Did you want me to do anything, or will the standard zipfiles/tarfiles you can get from Github work?

Thanks
Ian

HadetTheUndying, 2021-11-01:
Version numbered Source archives (tar & zip ) is what I'm requesting. So yes if that's possible for you under releases it would help maintainers off packages in non- bleeding edge distributions package this.

Thank you.

jayzeng, 2021-11-09:
+1 on this. @IanHarvey  will be awesome if you can publish the latest to pypi and/or other distro packages