IanHarvey/bluepy #466: How to use writeCharateristic() method by UUID?

Lance-Tin, 2021-11-28:
Thank you for your sharing of this respository. 
I know that the peripheral class have the method of ```writeCharateristic()``` and ```readCharateristic()```, But the two methods only can use handle to communicate with the peripherals, sometimes the handle is not useful and I can't get the True handle and UUID whith the method of ```getCharateristic()``` .

doug-holtsinger, 2021-12-16:
Hello,
You should be able to use the getHandle() method of the Characteristic() class to retrieve the Value Handle.  Alternatively you can simply use the write() and read() methods of the Characteristic() class directly and not use a handle.
http://ianharvey.github.io/bluepy-doc/characteristic.html

Lance-Tin, 2021-12-17:
Thank you for your answer, It's really a easy way to use "UUID" to communicate with the BLE peripheral, I will try again.

http://ianharvey.github.io/bluepy-doc/characteristic.html