IanHarvey/bluepy #467: blescan fails with errors on raspberry pi 4 running buster

americanium, 2021-12-17:
Hello,

running "blescan" the following error messages are displayed:

Scanning for devices...
Traceback (most recent call last):
  File "/usr/local/bin/blescan", line 33, in <module>
    sys.exit(load_entry_point('bluepy==1.3.0', 'console_scripts', 'blescan')())
  File "/usr/local/lib/python3.7/dist-packages/bluepy/blescan.py", line 122, in main
    devices = scanner.scan(arg.timeout)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 854, in scan
    self.stop()
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 803, in stop
    self._mgmtCmd(self._cmd()+"end")
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 312, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 11, error: Rejected)

Can anyone help me ?

Thank you!

doug-holtsinger, 2021-12-17:
Try running it with 'sudo' before the command.

doug-holtsinger, 2021-12-17:
I can re-create that error on my Raspberry Pi 3 if I disable the BLE with the "power off" command before running the blescan.  Check to see that your BLE is enabled.  You can do this in bluetoothctl:

pi@raspberrypi:~ $ bluetoothctl
Agent registered
[bluetooth]# power on
Changing power on succeeded
[bluetooth]#

americanium, 2021-12-17:
Just tried, same error.

doug-holtsinger, 2021-12-17:
Try this, in addition to the other commands I mentioned:

sudo systemctl enable bluetooth.service
sudo systemctl start bluetooth.service
sudo systemctl status bluetooth.service
# In the corner of the GUI, there should be a menu with a Bluetooth symbol.  Click on that and
# make sure that Bluetooth is Turned On

I get the same "Rejected" error if I turn off any one of the three items above.

americanium, 2021-12-17:
Hi!
I have no linux GUI.

When i try your commands i get the following error when using the last command:

Dec 17 10:48:26 loxberry-database systemd[1]: Starting Bluetooth service...
Dec 17 10:48:26 loxberry-database bluetoothd[485]: Bluetooth daemon 5.50
Dec 17 10:48:27 loxberry-database bluetoothd[485]: Starting SDP server
Dec 17 10:48:27 loxberry-database bluetoothd[485]: Bluetooth management interface 1.14 initialized
Dec 17 10:48:27 loxberry-database bluetoothd[485]: Sap driver initialization failed.
Dec 17 10:48:27 loxberry-database bluetoothd[485]: sap-server: Operation not permitted (1)
Dec 17 10:48:27 loxberry-database systemd[1]: Started Bluetooth service.

doug-holtsinger, 2021-12-17:
What output do you get with this command:

gdbus introspect -y -d "org.bluez" -o "/org/bluez/hci0"

Specifically what does this yield, if you're unable to post all the output:

gdbus introspect -y -d "org.bluez" -o "/org/bluez/hci0" | grep Powered

americanium, 2021-12-17:
readwrite b Powered = true;

doug-holtsinger, 2021-12-17:
OK, that Powered is the equivalent of the GUI enable.
Let's try it the hard way.  In a separate window, run:
   sudo btmon
Then run blescan and check for any Red text highlights and any obvious errors in the btmon output.

americanium, 2021-12-17:
Bluetooth monitor ver 5.50
= Note: Linux version 4.19.118-v7+ (armv7l)                                                                                                                                                                                         0.053171
= Note: Bluetooth subsystem version 2.22                                                                                                                                                                                            0.053183
= New Index: B8:27:EB:03:45:86 (Primary,UART,hci0)                                                                                                                                                                           [hci0] 0.053187
= Open Index: B8:27:EB:03:45:86                                                                                                                                                                                              [hci0] 0.053189
= Index Info: B8:27:EB:03:45:86 (Broadcom Corporation)                                                                                                                                                                       [hci0] 0.053192
@ MGMT Open: bluetoothd (privileged) version 1.14                                                                                                                                                                          {0x0001} 0.053199
@ MGMT Open: btmon (privileged) version 1.14                                                                                                                                                                               {0x0002} 0.053287
@ MGMT Open: bluepy-helper version 1.14                                                                                                                                                                                   {0x0003} 27.730203
@ MGMT Command: Read Management Version Information (0x0001) plen 0                                                                                                                                                       {0x0003} 27.731099
@ MGMT Event: Command Complete (0x0001) plen 6                                                                                                                                                                            {0x0003} 27.731121
      Read Management Version Information (0x0001) plen 3
        Status: Success (0x00)
        Version: 1.14
@ MGMT Command: Set Low Energy (0x000d) plen 1                                                                                                                                                                     {0x0003} [hci0] 27.732052
        Low Energy: Enabled (0x01)
@ MGMT Event: Command Status (0x0002) plen 3                                                                                                                                                                       {0x0003} [hci0] 27.732068
      Set Low Energy (0x000d)
        Status: Permission Denied (0x14)
@ MGMT Close: bluepy-helper                                                                                                                                                                                               {0x0003} 27.732954
@ MGMT Open: bluepy-helper (privileged) version 1.14                                                                                                                                                                      {0x0003} 48.393797
@ MGMT Command: Read Management Version Information (0x0001) plen 0                                                                                                                                                       {0x0003} 48.394220
@ MGMT Event: Command Complete (0x0001) plen 6                                                                                                                                                                            {0x0003} 48.394242
      Read Management Version Information (0x0001) plen 3
        Status: Success (0x00)
        Version: 1.14
@ MGMT Command: Set Low Energy (0x000d) plen 1                                                                                                                                                                     {0x0003} [hci0] 48.394628
        Low Energy: Enabled (0x01)
@ MGMT Event: Command Complete (0x0001) plen 7                                                                                                                                                                     {0x0003} [hci0] 48.394653
      Set Low Energy (0x000d) plen 4
        Status: Success (0x00)
        Current settings: 0x00000ad1
          Powered
          Bondable
          Secure Simple Pairing
          BR/EDR
          Low Energy
          Secure Connections
@ MGMT Command: Start Discovery (0x0023) plen 1                                                                                                                                                                    {0x0003} [hci0] 48.396140
        Address type: 0x06
          LE Public
          LE Random
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                                                                                                                                                                #1 [hci0] 48.396460
        Address: 0F:AA:3E:42:57:92 (Non-Resolvable)
[... cut ...]
@ MGMT Command: Stop Discovery (0x0024) plen 1                                                                                                                                                                     {0x0003} [hci0] 54.443850
        Address type: 0x06
          LE Public
          LE Random
@ MGMT Event: Command Complete (0x0001) plen 4                                                                                                                                                                     {0x0003} [hci0] 54.443884
      Stop Discovery (0x0024) plen 1
        Status: Rejected (0x0b)
        Address type: 0x06
          LE Public
          LE Random

americanium, 2021-12-17:
Sorry, i didn't want to close this issue ...

@ MGMT Event: Command Status (0x0002) plen 3                                                                                                                                                                       {0x0003} [hci0] 27.732068
      Set Low Energy (0x000d)
        Status: Permission Denied (0x14)

PERMISSION DENIED = RED

@ MGMT Event: Command Complete (0x0001) plen 4                                                                                                                                                                     {0x0003} [hci0] 50.441895
      Start Discovery (0x0023) plen 1
        Status: Failed (0x03)

FAILED = RED

@ MGMT Event: Command Complete (0x0001) plen 4                                                                                                                                                                     {0x0003} [hci0] 54.443884
      Stop Discovery (0x0024) plen 1
        Status: Rejected (0x0b)
        Address type: 0x06
          LE Public
REJECTED = RED.

doug-holtsinger, 2021-12-17:
That's strange.  I captured your full output before you edited it (?), and I see that it ran the "Low Energy On" command twice, the first time it failed with a permission error and the second time it passed.  It should always either fail or pass, and I can't see how the command could be called twice from bluepy.

You can run the following commands to simplify the debug of the two failing commands in your output (Set Low Energy and Start Discovery), and see if you can find a pattern of failures, and continue to monitor the output of btmon:

pi@raspberrypi:~ $ sudo btmgmt
[mgmt]# le on
hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn
[mgmt]# le on
hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn
[mgmt]# find
//  exit when you are done

You might also try running blescan as root, but I doubt that will change the result.

americanium, 2021-12-18:
[mgmt]# le on
hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn
[mgmt]# le on
hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn
[mgmt]# find
Unable to start discovery. status 0x03 (Failed)

doug-holtsinger, 2021-12-20:
Could you please capture the output of this series of commands? 

export MGMT_DEBUG=1
sudo btmgmt
power on
le on
le on
info
find
version
exit
dmesg | tail -10
uname -a

americanium, 2021-12-20:
root@loxberry-database:/opt/loxberry# export MGMT_DEBUG=1
root@loxberry-database:/opt/loxberry# sudo btmgmt
[mgmt]# power on
hci0 Set Powered complete, settings: powered bondable ssp br/edr le secure-conn
[mgmt]# le on
hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn
[mgmt]# le on
hci0 Set Low Energy complete, settings: powered bondable ssp br/edr le secure-conn
[mgmt]# info
Index list with 1 item
hci0:   Primary controller
        addr B8:27:EB:03:45:86 version 7 manufacturer 15 class 0x000000
        supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy static-addr
        current settings: powered bondable ssp br/edr le secure-conn
        name loxberry-database
        short name
[mgmt]# find
Unable to start discovery. status 0x03 (Failed)
[mgmt]# version
Version 5.50
[mgmt]# exit
root@loxberry-database:/opt/loxberry# dmesg | tail -10
[246276.509899] systemd-fstab-generator[31942]: Ignoring "nofail" for root device
[249877.032694] systemd-fstab-generator[1514]: Ignoring "nofail" for root device
[253476.478155] systemd-fstab-generator[3782]: Ignoring "nofail" for root device
[257077.406269] systemd-fstab-generator[5758]: Ignoring "nofail" for root device
[260676.618581] systemd-fstab-generator[7745]: Ignoring "nofail" for root device
[264277.243975] systemd-fstab-generator[9723]: Ignoring "nofail" for root device
[267876.486044] systemd-fstab-generator[11709]: Ignoring "nofail" for root device
[270541.306015] Bluetooth: hci0: command 0x2005 tx timeout
[270543.386009] Bluetooth: hci0: command 0x200b tx timeout
[270545.466004] Bluetooth: hci0: command 0x200c tx timeout
root@loxberry-database:/opt/loxberry# uname -a
Linux loxberry-database 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
root@loxberry-database:/opt/loxberry#

americanium, 2021-12-20:
Now it works!
I installed / updated the following packages:
python python-pip python-dev python-bluez raspberrypi-sys-mods pi-bluetooth bluez