IanHarvey/bluepy #477: bluepy.btle.BTLEManagementError: Failed to execute management command 'le on' (code: 17, error: Invalid Index)

Ajith61, 2022-04-21:
bluepy.btle.BTLEManagementError: Failed to execute management command 'le on' (code: 17, error: Invalid Index)

maciej-napora, 2022-05-17:
Does this help? https://github.com/IanHarvey/bluepy/issues/313

myfingerhurt, 2022-07-21:
@maciej-napora 
I have the same issue, with `sudo` doesn't work either.

blshkv, 2024-06-07:
I face the same problem. According to the solution, you need to install `pulseaudio-module-bluetooth``` module.
However, I updated my system from pulseaudio to pipewire recently, so that module is no longer relevant.
I double checked, pipewire does has bluetooth module enabled:
```
 media-video/pipewire-1.0.5:0/0.4::gentoo  USE="X bluetooth dbus echo-cancel extra ffmpeg lv2 readline sound-server ssl v4l -doc -flatpak -gsettings -gstreamer -ieee1394 -jack-client -jack-sdk -liblc3 -man -modemmanager -pipewire-alsa -roc (-selinux) (-system-service) -systemd -test -zeroconf" 
```