IanHarvey/bluepy #487: Sampling frequency

goru89, 2022-06-22:
Hi. I am working on a CC2650 Sensor Tag and I am able to fetch the accelerometer data. But the sampling frequency seems to be slow for my purpose. How can I control the sampling frequency to 50Hz using the functions.

maciej-napora, 2022-06-23:
That has little to do with bluepy.

 Sampling frequency of accelerometer IC is one thing. How often you are notified by Bluetooth Low Energy stack is another thing. I assume you want to receive notifications more often. Look in the documentation for how to set the frequency of notifications.

goru89, 2022-06-23:
Hmm. Its like. Currently the Sensor is reading 10 values per second which is default. But is there a way to make it 50Hz. When I checked the manual, they have mentioned that we have to write the period to 0xAA83. In bluepy.sensortag do we have any inbulit methods to do that ?
[image]

maciej-napora, 2022-06-23:
You can use general functions _getServiceByUUID_ to retrieve the service from peripheral, then use _getCharacteristics_ with UUID AA83* (ofc providing full UUID) and then simply write to this characteristic (_write_) to set the period. 

Unfortunately, I see from what copy-pasted that minimum period is 100 ms, so you will not be able to be notified 50 Hz. How often the data is sampled in SensorTag is a different thing. This seems to be a limitation of SensorTag firmware.

goru89, 2022-06-23:
Oh right. So that would mean 10Hz is the max sampling rate. Right ?

maciej-napora, 2022-06-23:
It seems so. I know that MPU9250 is able to sample data much faster, but there is probably a while loop somewhere in the firmware for which you set the period writing to AA83 characteristic. The designers for various reasons might have chosen to set the minimum period of this loop to 100 ms (10 Hz).

goru89, 2022-06-23:
Hmm. Makes Sense. Cool . Thanks buddy.

closed 2022-06-23 by goru89