IanHarvey/bluepy #492: Try to use timeout

xav12358, 2022-07-19:
Hello,
I try to use connect function with a timeout. I use bluepy in 1.3.0 and it seems there is no timeout argument.

Si I try to design a timeout with multithread and queue.

Here is my code:
```

       #####################################"
        # def connection_worker(ret_status_queue, new_num_list_in):
        def connection_worker(ret_status_queue, ret_service_queue):

            try:
                self.peripheral.connect(ble_mac_address)
                time.sleep(0.1)
                service = self.peripheral.getServiceByUUID(self.SERVICE_UUID)
                characteristic = service.getCharacteristics()[0]
                ret_characteristic_queue.put(characteristic )
                ret_status_queue.put("OK")
            except Exception as err:
                print('errin {}'.format(err))
                ret_status_queue.put("NOK")
                ret_status_queue.task_done()

        ret_status_queue = multiprocessing.Queue()
        ret_characteristic_queue = multiprocessing.Queue()
        # proc = multiprocessing.Process(target=connection_worker, args=(ret_status_queue,new_num_list))
        proc = multiprocessing.Process(target=connection_worker, args=(ret_status_queue,ret_characteristic_queue))
        proc.start()
        try:
            ret_value = ret_status_queue.get(timeout=22)
        except Exception as err:
            ret_value = 'NOK'
        proc.terminate()

        if ret_value == 'OK':
            # Get characteristic
            try:
                characteristic = ret_service_queue.get(timeout=5) // Not good characterictic
                self.characteristic = characteristic
            except Exception as err:
                print('holala {}'.format(err))
```

What is the proper way to set a timeout function on connection?

maciej-napora, 2022-07-19:
This repo in the latest version already contains connect() with a timeout,
look at the source code in this repo.

XavierExceenis, 2022-07-19:
I use the last tag 1.3.0. It is not inside that release ?

XavierExceenis, 2022-07-19:
When should a new tag generate to integrate that modification?

xav12358, 2022-07-19:
When should a new tag generate to integrate that modification?

maciej-napora, 2022-07-19:
My feeling is that this should take a while, as there is not much action happening here. You can always download the most recent code from this repo and call it from your application, instead of the lib installed by PIP. That's what I have been doing.