IanHarvey/bluepy #496: Select a host Bluetooth device (the one that starts the connection)

toncho11, 2022-10-20:
Hi,

Is it possible to select between multiple USB plugged **host** Bluetooth devices? On my computer I have several Bluetooth devices and I want to initiate a connection from a specific one towards another Bluetooth device. How can I do that? 

I think devices are enumerated: `hci0` , `hci1` ...
I do not want to shutdown the not used device, just select another one.

toncho11, 2022-10-25:
I think I found it:

http://ianharvey.github.io/bluepy-doc/peripheral.html?highlight=peripheral

In the class Peripheral we can select which device we would like to use: `On Linux, 0 means /dev/hci0, 1 means /dev/hci1 and so on.`

closed 2022-10-25 by toncho11

kg-meso, 2023-02-13:
Did this work for you? I am running bluepy on Raspbian and there is no /dev/hciX so setting the iface value does not affect which bluetooth device initiates the connection. See my issue #501. If you got it to work, do you mind sharing what OS you are using?

toncho11, 2023-02-13:
I got it running on Linux, actually on WSL2 (which requires a lot of extra effort).
I modified https://github.com/toncho11/pyparrot to use more than 1 Bluetooth device. 
You need to search of hci devices with `dmesg` command. Kernel messages about hci devices should appear when you plug them in. Your problem seems specific to RasberryPi.

kg-meso, 2023-02-13:
Interesting, I am able to see both of these devices as hci devices with specific _hciX_ id's. Perhaps this codebase does not work well with Raspbian? If you call the command '_ls /dev/hci*_' do you actually see _hciX_ id's?

toncho11, 2023-02-13:
Sorry I do not have access right now.

kg-meso, 2023-02-13:
No problem, can you recall what distro you were using? I was thinking of trying this code on Ubuntu.

toncho11, 2023-02-13:
I use WSL2 (Windows Subsystem for Linux), but you need many things to get Bluetooth working on it.
Ubuntu should be OK.