IanHarvey/bluepy #507: Bluepy and Bookworm

madmacks59, 2023-11-24:
I've been running Bluepy on several Bullseye based RPis for awhile. Recently I had to reformat the SD card on one of those Pis, and I decided to upgrade to Bookworm, as it's the latest Pi OS. All went well, except now I can't install Bluepy on the Pi. When I run the "normal" install process the OS throws an error...

sudo pip install bluepy

results in:
error: externally-managed-environment

This environment is externally managed
To install Python packages system-wide, try apt install python3-xyz, where xyz is the package you are trying to install.

So I try...

sudo apt install python3-bluepy

and the response is...

E: Unable to locate package python3-bluepy

Is there some trick to getting Bluepy  installed on Bookworm?

dheuts90, 2024-01-12:
Same question over here... FOund any solutions?

theyosh, 2024-01-26:
I would go for: https://pypi.org/project/bluepy3/ It should be based on this code. But for python >=3.8 only

OnlyNandan, 2024-03-08:
Im having the same issue. Using it for wof. Did you guys have any luck?

madmacks59, 2024-03-08:
I had to set up and use a VENV using Bookworm. Once I figured out how to do that it worked fine, but it took me awhile to figure it out.

OnlyNandan, 2024-03-09:
So all I have to do is get bluepy installed with venv `using source` and just run and check if wof program is running then deactivate and then install the plugin

technomav, 2024-04-09:
Add this at the end of pip install "--break-system-packages"
eg: sudo pip install bluepy --break-system-packages
This will install bluepy as its install earlier, no need to change any libs. 

This is due to your distribution adopting [PEP 668 – Marking Python base environments as “externally managed”](https://peps.python.org/pep-0668/).